LazeMSS/OctoPrint-TopTemp #105: Small cosmetic Issue

mike-bee, 2024-02-05:
I am running all latest stable versions and have a small issue with the TopTemp appearance: There is a unreasonable big gap between the icon and the readout. At least it is bigger than on the image in your ReadMe on the Code tab.

[image]

LazeMSS, 2024-02-05:
Will look when back from vacation

LazeMSS, 2024-02-12:
Maybe its a theme - do you have the same problem with all themes?

mike-bee, 2024-02-12:
Welcome back - hope you had a good time.
I guess you refer to "UI Customizer" themes. Yes it's the same in all. Plaese see Default theme:
[image]

LazeMSS, 2024-02-12:
I can't recreate it - might there might another plugin running etc. please post screenshot of entire top bar

mike-bee, 2024-02-12:
[image]
[image]
[image]

LazeMSS, 2024-02-12:
Please check you don't have your browser zoomed in in font size

mike-bee, 2024-02-12:
Chrome zoom level is at 100%. Though, I tested zooming in and out. The issue disappeared from 175% upwards:
[image]

LazeMSS, 2024-02-12:
Maybe an overflow issue with the printer label etc. - seems strange nobody else has this issue :/

mike-bee, 2024-02-13:
Too bad. I can try changing my settings and watch for improvement. Anything you recommend to start with or focus on?

mike-bee, 2024-02-14:
**Good news**: I was able to recover the look-and-feel. I started gradually removing labels first, then complete items. No change.
Then uninstalled (including data) and reinstalled Top Temp and since the original look-and-feel is back. GREAT. I reconfigured only the bare minimum. Still looking good.
**Bad news**: No chance of analyzing the issue any further.

closed 2024-02-14 by mike-bee